p3p/pio-framework-arduino-lpc176x #9: Feature Request: would it be possible to implement micros()?

andyman198, 2018-11-07:
If it is acceptable to ask for feature requests, I would like to ask for micros(). I am trying to look at what needs to be implemented to get neopixels working. I am not sure millis()*1000 would be enough. https://www.arduino.cc/reference/en/language/functions/time/micros/

p3p, 2018-11-07:
Of course it's acceptable, especially when its a missing part of the Arduino API, same with #10, but as for 3rd party library support most bypass the Arduino API and access AVR function directly so it will be unlikely we will ever support many without having a specific port, I only had a quick look at neopixel but it seemed to be platform specific.

andyman198, 2018-11-07:
wonderful :) thank you for the feedback. I realize the adafruit library will need a specific implementation so I will look at that, but I figured if I see things in passing that would be helpful to complete this library then would be good to share. I am appreciative that you are open to adding them in, it will undoubtedly help with others as they connect libraries in.

closed 2018-11-20 by p3p