p3p/pio-framework-arduino-lpc176x #20: TMC CONNECTION ERROR with TMC2660 and LPC1769

andyman198, 2019-04-30:
### Description

Not sure if this is an issue with the SPI or the TMC library so close if its out of place. I am having issues with all the TMC2660 drivers in marlin. 

M122 Below -------

```
SENDING:M122
		X	Y	Z	E	E1
Enabled		true	true	true	true	true
Set current	800	800	800	800	800
RMS current	673	673	673	673	673
MAX current	949	949	949	949	949
Run current	23/31	23/31	23/31	23/31	23/31
Hold current
CS actual
PWM scale
vsense		1=.165	1=.165	1=.165	1=.165	1=.165
stealthChop
msteps		16	16	16	16	16
tstep
pwm
threshold
[mm/s]
OT prewarn
OT prewarn has
been triggered
off time		3	3	3	3	3
blank time	24	24	24	24	24
hysteresis
-end		-1	-1	-1	-1	-1
-start		1	1	1	1	1
Stallguard thrs	0	0	0	0	0
DRVSTATUS	X	Y	Z	E	E1
stst
olb
ola
s2gb
s2ga
otpw
ot
Driver registers:
		X	0x00:00:00:00	 Bad response!
		Y	0x00:00:00:00	 Bad response!
		Z	0x00:00:00:00	 Bad response!
		E	0x00:00:00:00	 Bad response!
		E1	0x00:00:00:00	 Bad response!
Testing X connection... Error: All LOW
Testing Y connection... Error: All LOW
Testing Z connection... Error: All LOW
Testing E connection... Error: All LOW
Testing E1 connection... Error: All LOW
```

### Steps to Reproduce

   #define MOTHERBOARD BOARD_AZTEEG_X5_GT
   #define X_DRIVER_TYPE  TMC2660
   #define Y_DRIVER_TYPE  TMC2660
   #define Z_DRIVER_TYPE  TMC2660
   #define E0_DRIVER_TYPE TMC2660
   #define E1_DRIVER_TYPE TMC2660

**Expected behavior:** [What you expect to happen]
   Can see TMC drivers and configure

**Actual behavior:** [What actually happens]
  TMC CONNECTION ERROR on LCD and Bad Response! on M122

p3p, 2019-04-30:
You're more likely to get help on a forum or if it is a bug on the Marlin repository issues, I don't control all the library and SPI interactions from the framework that's upto Marlin.

andyman198, 2019-04-30:
Ok. Thanks for letting me know. I know this worked before and wanted to see if there was any framework level SPI changes that could have affected it. I will close the issue.

closed 2019-04-30 by andyman198