p3p/pio-framework-arduino-lpc176x #26: Multiple definition of eeprom functions when compiling Marlin 2 with framework-arduino-lpc176x@0.2.1

atoomnetmarc, 2019-11-09:
Hi,

When compiling for a board with a LPC1768 you get a multiple definition error on the eeprom functions at the linking stage. The empty eeprom functions in arduino.cpp conflict with eeprom_i2c.cpp from Marlin 2.

Config:
#define I2C_EEPROM
#define E2END 0x7FFF
#define MOTHERBOARD BOARD_BIGTREE_SKR_V1_3

First error:
Linking .pio\build\LPC1768\firmware.elf
c:/users/marc/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio\build\LPC1768\libFrameworkArduino.a(arduino.cpp.o): in function `eeprom_write_byte(unsigned char*, unsigned char)':
C:\Users\Marc\.platformio\packages\framework-arduino-lpc176x\cores\arduino/arduino.cpp:112: multiple definition of `eeprom_write_byte(unsigned char*, unsigned char)'; .pio/build/LPC1768/src/src/HAL/shared/eeprom_i2c.cpp.o:e:\Marlin-bugfix-2.0.x/Marlin\src\HAL\shared/eeprom_i2c.cpp:49: first defined here

When I comment out the empty eeprom functions in arduino.cpp the compilation succeeds.

/*
void eeprom_write_byte(uint8_t *pos, unsigned char value) { }

uint8_t eeprom_read_byte(uint8_t * pos) { return '\0'; }

void eeprom_read_block(void *__dst, const void *__src, size_t __n) { }

void eeprom_update_block(const void *__src, void *__dst, size_t __n) { }
*/

Is this issue something that may be fixed in pio-framework-arduino-lpc176x?

p3p, 2019-11-09:
> Is this issue something that may be fixed in pio-framework-arduino-lpc176x?

The lpc1768/9 do not have an inbuilt EEPROM so these functions can't be defined, but some other lpc176x compatible mcus that may be supported in the future do have an eeprom.. 

I'm not sure the correct approach as these function are not necessarily part of the Arduino api but they do seem do be used as such.

The correct fix for this is Marlins eeprom i2c implementation been updated to use the persistent store api as it should.

atoomnetmarc, 2019-11-10:
OK, thanks for the pointer.

closed 2019-11-10 by atoomnetmarc