p3p/pio-framework-arduino-lpc176x #27: Ability to use alternative UART ports

msutas, 2019-11-17:
Is it possible to define the uart ports via the Marlin configuration. For example by default uart3 is binded to P0_00 - P0_01 but I would like to use P4_28 - P4_29 pins.

I can do this by modifying the HardwareSerial.h file that is downloaded by platformio but I would like to add my board files to Marlin source and since the uart pins are hard coded my uart3 will not be functional in Marlin for someone using the board setup other than me.

I thought this is the right place to ask rather than Marlin repository.

p3p, 2019-11-17:
Could probably just change the constructor to take pins, or set an alt mode, would just need to update Marlins MarlinSerial class to use it.

msutas, 2019-11-18:
Thank you let me dig it. If I can solve it, I wil create pull requests on Marlin and the framework.

closed 2019-11-18 by msutas