p3p/pio-framework-arduino-lpc176x #42: PlatformIO Core 4.4 and the package registry

ivankravets, 2020-08-17:
Hi Chris,

Thank you so much for your work and that use PlatformIO.

We see you use a legacy `packageRepositories` at https://github.com/p3p/pio-nxplpc-arduino-lpc176x/blob/release/platform.json#L17

This feature has been removed in the upcoming PlatformIO Core 4.4. We uploaded `framework-arduino-lpc176x @ 0.2.4` to the registry.

Would you like to maintain this package later by yourself? There is a new command `pio package publish` in PIO Core 4.4. We can also transfer current `framework-arduino-lpc176x @ 0.2.4` to your PlatformIO Account.

If you have any questions, do not hesitate to contact us.

ivankravets, 2020-09-06:
Sorry for not replying earlier. Release weeks are always tough.

> whats the reason you are deprecating this feature?

There are a lot of reasons, see https://docs.platformio.org/en/latest/core/migration.html#package-management
Mostly, platform maintainers put packages to their servers which are not accessible for other devs. The same when putting packages on Github. Github is blocked in China. So, developers in this case attack us. They think that the problem is with PlatformIO. The solution to this problem is our own trusted registry and whole infrastructure which we built.

We allow anyone to publish a package to our registry and we will guarantee availability for it.

> Would you like to maintain this package later by yourself

1. Please merge https://github.com/p3p/pio-framework-arduino-lpc176x/pull/44
2. We've moved this package to your PlatformIO Account `p3p`. If you forgot a password, please restore it.
3. If you update this package, just increment the version, change the folder to the root of the package in your terminal and type `pio package publish`.

- https://docs.platformio.org/en/latest/plus/pio-account.html
- https://docs.platformio.org/en/latest/core/userguide/package/cmd_publish.html#cmd-package-publish

p3p, 2020-09-06:
I can see the fragmentation being an issue, especially with you getting the complaints rather than the platform maintainers, I guess the only downside is now if your infrastructure goes down rather than githubs the builds will fail ^^. Not sure how Chinese devs handle github being blocked, can't imagine the amount of things that breaks.

Thanks for moving the package, is there a command to list the packages on my account?

ivankravets, 2020-09-07:
> Not sure how Chinese devs handle github being blocked, can't imagine the amount of things that breaks.

See https://github.com/Nuclei-Software/platforma-nuclei/issues/7 . A few hours ago.

>  is there a command to list the packages on my account?

Yes, you need https://docs.platformio.org/en/latest/core/userguide/access/cmd_list.html

Does it show your packages?

p3p, 2020-09-07:
Indeed `pio access list` is what I was looking for
```
U8glib-ARM
----------
URN:              prn:reg:pkg:1691:library
Owner:            p3p
Access level(s):  Admin

framework-arduino-lpc176x
-------------------------
URN:              prn:reg:pkg:11002:tool
Owner:            p3p
Access level(s):  Admin
```
The description of the `access` command must have put me off looking at it closer, expected it under the `package`.

ivankravets, 2020-09-07:
Could we improve something?

p3p, 2020-09-07:
It may be a good idea to have a way to list account packages under the package command, but mostly it was just me not reading the docs well enough and the help in the `pio` command was just `Manage resource access` which didn't click to me as a way to list packages.

ivankravets, 2020-09-07:
Thanks! Fixed https://github.com/platformio/platformio-docs/commit/30ff70a4cef16890868cd95e4a68f75cae1b861d

So, I close this issue. Please don't hesitate to contact us for any questions.

closed 2020-09-07 by ivankravets